juliansteenbakker/flutter_ble_peripheral #26: Working on my Senior project and need help please.

Mrsinan9, 2020-11-21:
I need to connect two devices using BLE. and I have successfully connected both devices using your library and flutter blue library.
now I want to read stuff from the peripheral and want to create custom services .how can do that? whenever i add service uuid to the the data an error will appear saying it is too large. 
please help us cause this is s core feature in our project. thanks

Mrsinan9, 2020-11-22:
Also could you please provide sample code for custom services and characteristics for android ?

Mrsinan9, 2020-11-22:
@juliansteenbakker

juliansteenbakker, 2020-11-24:
The library currently only supports broadcasting service UUIDs which can be 16 bits long. (e.g. List<int>(16);)
I'll look what I can do regarding reading data however Im currently quite busy so I cant guarantee anything. Feel free to add features and create a pull request if you have any ideas. 

Broadcasting a custom service and characteristic is currently also not possible. I've updated the readme.

juliansteenbakker, 2021-04-06:
I have added possibility to advertise custom service UUID in Android. This is not possible in iOS.

closed 2021-04-06 by juliansteenbakker

RazaGR, 2021-06-05:
@juliansteenbakker I was reading your article on dev.to where you mentioned:

> I’ve developed a way to broadcast custom data using the UUID field of the Core Bluetooth API. 

I wanted to ask, how do you do that? do you set the data you need inside UUID?
in this case, could we update UUID dynamically while advertising ? so this way we can get data from UUID dynamically.