juliansteenbakker/flutter_ble_peripheral #27: Send data from flutter ble aplication to ble device with button click

Ibrahim23123, 2021-01-06:
In my project, I want to use a button click for send data from ble application to ble device but I could not send data with a button click.In my code using "characteristic.write" method to send data and so this method use only if you write data and then application send data to ble device so how can I convert write data method for only button click without write any data. For example when I click a button and then my ble device's led state is on. And here partial of sending data in my code :   

    if (characteristic.properties.write) {
      buttons.add(
        ButtonTheme(
          minWidth: 10,
          height: 20,
          child: Padding(
            padding: const EdgeInsets.symmetric(horizontal: 4),
            child: RaisedButton(
              child: Text('YAZ', style: TextStyle(color: Colors.white)),
              onPressed: () async {
                await showDialog(
                    context: context,
                    builder: (BuildContext context) {
                      return AlertDialog(
                        title: Text("Yaz"),
                        content: Row(
                          children: <Widget>[
                            Expanded(
                              child: TextField(
                                controller: _writeController,
                              ),
                            ),
                          ],
                        ),
                        actions: <Widget>[
                          FlatButton(
                            child: Text("Gönder"),
                            onPressed: () {
                              characteristic.write(
                                  utf8.encode(_writeController.value.text));
                              Navigator.pop(context);
                            },
                          ),
                          FlatButton(
                            child: Text("İptal"),
                            onPressed: () {
                              Navigator.pop(context);
                            },
                          ),
                        ],
                      );
                    });
              },
            ),
          ),
        ),
      );
    }

juliansteenbakker, 2021-01-06:
Are you using an Android or iOS device?

Ibrahim23123, 2021-01-06:
Android

Ibrahim23123, 2021-01-06:
Android

Julian Steenbakker <notifications@github.com>, 6 Oca 2021 Çar, 12:16
tarihinde şunu yazdı:

juliansteenbakker, 2021-01-06:
I have an update pending for android that i will release today. Please stay tuned. I'll also add some documentation. In short; on Android it will be possible to advertise manufacturerData, serviceData and a self set UUID. On iOS, it wil only be possible to advertise a self set UUID since apple prohibits the use of manufacturerData and serviceData.

juliansteenbakker, 2021-04-06:
Released v0.4.0

closed 2021-04-06 by juliansteenbakker