juliansteenbakker/flutter_ble_peripheral #37: Flutter: Swift Compiler Error in v0.4.0

shobhit14gupta, 2021-07-13:
While upgrading flutter_ble_peripheral to v0.4.0, I encountered an error during the build process of the application on iOS. The following error shows on Peripheral.swift file:
1.  Immutable value 'self.peripheralManager' may only be initialized once.
2. Property 'self.peripheralManager' not initialized at super.init call.

Kindly help me with it! As I would like to advertise deviceName on iOS device. 

[image]

juliansteenbakker, 2021-07-14:
Please try again with v0.4.1

closed 2021-07-14 by juliansteenbakker

shobhit14gupta, 2021-07-14:
Thank you for reply, julian! The above error is sloved now. But, still iOS device does not able to advertise the deviceName.  
Kindly check the mapping parameters in SwiftFlutterBlePeripheralPlugin.swift. I think there is syntax error with the map key  "deviceName" should be there instead of "localName".  Kindly look forward to it.  Thank you.

shobhit14gupta, 2021-07-14:
> Please try again with v0.4.1

Thank you for reply, julian! The above error is sloved now. But, still iOS device does not able to advertise the deviceName.
Kindly check the mapping parameters in SwiftFlutterBlePeripheralPlugin.swift. I think there is syntax error with the map key "deviceName" should be there instead of "localName". Kindly look forward to it. Thank you.

juliansteenbakker, 2021-07-14:
Thanks for noticing! I released v0.4.2 with fixed typo.

shobhit14gupta, 2021-07-15:
> Thanks for noticing! I released v0.4.2 with fixed typo

Still not possible to advertise local Name in iOS devices. After debugging, I have found the local Name remains Nil. Possibly, the deviceName data type (List<int>) not able to convert into String data type. Moreover, I would suggest to change the CB AdvertisementDataLocalNameKey into String data type instead of passing to the CBUUID.

Kindly look forward to it. Thank You.