juliansteenbakker/flutter_ble_peripheral #41: Android Build Failed in V0.5.0+1.

shobhit14gupta, 2021-07-28:
Hello, I got the build error during android build of the flutter application.  The error state :  

"Module was ccompiled with an incompatible version of Kotlin. The binary version of its metadata is 1.5.1, expected version is 1.1.15."

The plugin was working earlier with version 0.4.0.

Here is the info from command  flutter doctor -v:

Flutter (Channel stable, 2.2.3, on Microsoft Windows [Version 10.0.19043.1110], locale de-DE)
    • Flutter version 2.2.3 
    • Framework revision f4abaa0735 (4 weeks ago), 2021-07-01 12:46:11 -0700
    • Engine revision 241c87ad80
    • Dart version 2.13.4

[√] Android toolchain - develop for Android devices (Android SDK version 30.0.3)
    • Platform android-30, build-tools 30.0.3
    • Java version OpenJDK Runtime Environment (build 11.0.8+10-b944.6842174)   
    • All Android licenses accepted.

[√] Chrome - develop for the web
    • Chrome at C:\Program Files\Google\Chrome\Application\chrome.exe

[√] Android Studio (version 4.1.0)
    • Android Studio at C:\Program Files\Android\Android Studio
    • Flutter plugin can be installed from:
       https://plugins.jetbrains.com/plugin/9212-flutter
    • Dart plugin can be installed from:
       https://plugins.jetbrains.com/plugin/6351-dart
    • Java version OpenJDK Runtime Environment (build 11.0.8+10-b944.6842174)

[image]

juliansteenbakker, 2021-07-28:
can you please try the following:

In android/build.gradle change ext.kotlin_version = '1.3.50' to ext.kotlin_version = '1.5.10'

In android/build.gradle change classpath 'com.android.tools.build:gradle:3.5.0' to classpath 'com.android.tools.build:gradle:4.2.0'

In android/gradle/wrapper/gradle-wrapper.properties change distributionUrl=https\://services.gradle.org/distributions/gradle-5.6.2-all.zip to distributionUrl=https\://services.gradle.org/distributions/gradle-6.9-all.zip

shobhit14gupta, 2021-07-28:
Thank you. It works. :thumbsup:

closed 2021-07-28 by shobhit14gupta