juliansteenbakker/flutter_ble_peripheral #77: Peripheral get connected and let the central read and write the data

HaniMontana, 2022-04-30:
Hello guys,

currently I am trying to evaluate if its possible that my Android and iOS device is advertising as peripheral. And additional if its possible that a central device also a Andoird or IOS is able to connect to the peripheral read the internal data and also write to the internal data. There is of course a pairing required like "JustWorks" or "bonding". There is no flutter lib where this functionality is available and my question is: Is this possible?
In general when I read about the iOS Lib "Core bluetooth" it seem to me that this should be possible.

Best regards 
Hani

klipsedeville, 2024-02-05:
I need the same thing. Please did you get a solution?

juliansteenbakker, 2026-09-03:
It's been a while, but i have released a new version of this package, [v3.0.0](https://github.com/juliansteenbakker/flutter_ble_peripheral/releases/tag/v3.0.0), together with the other side of it, [flutter_ble_central](https://github.com/juliansteenbakker/flutter_ble_central/releases/tag/v1.0.0) v1.0.0. 3.0.0 adds a gatt server with tx and rx characteristics, so a central can connect and read and write over the link. The example app does exactly that, with flutter_ble_central on the other end.

Please check again with these versions, and reopen if it's still there.

closed 2026-09-03 by juliansteenbakker