juliansteenbakker/flutter_ble_peripheral #80: how to change android ble name?

Liberations, 2022-05-24:
localName seem don't work

fernandoaestrella, 2024-10-09:
changing my device name via Android's OS settings has changed the transmitted device name for me when using other packages

juliansteenbakker, 2026-09-03:
It's been a while, but i have released a new version of this package, [v3.0.0](https://github.com/juliansteenbakker/flutter_ble_peripheral/releases/tag/v3.0.0), together with the other side of it, [flutter_ble_central](https://github.com/juliansteenbakker/flutter_ble_central/releases/tag/v1.0.0) v1.0.0. Android does not broadcast `localName`, it broadcasts the adapter name, which an app cannot change. 3.0.0 only offers `localName` where the platform actually sends one, and on android you use `includeDeviceName` instead.

Please check again with these versions, and reopen if it's still there.

closed 2026-09-03 by juliansteenbakker