juliansteenbakker/flutter_ble_peripheral #85: GATT SERVER

SwapnaliInsiza, 2022-06-16:
Hi Guys,
This package does help in advertising the Android device as peripheral. Any help with building connection between the peripheral and host? Can we have the GATT services added within this package as that will extend the functionality within the package?

MrCsabaToth, 2022-12-30:
Now I'm looking at the code and I don't see how to communicate data via the advertised service. For me there wouldn't be reasonable to advertise something without being able to communicate. So it wouldn't just look like a duck but it would walk and quack like a duck.

MrCsabaToth, 2022-12-31:
Looks like there's a ton of commented TODO code related to `sendData` and setting up a server and characteristic. I tried to brush it up but my problem is that even the simple example (without sendData or anything) doesn't work. The advertised service doesn't shop up in any scans.

sjsilverstein, 2024-08-07:
It looks like development on this has come to a bit of a stand still. I see a pull request in draft for a GATT service code on android. Is there any intent to add this functionality to the package?

Luismi74, 2025-02-24:
> Looks like there's a ton of commented TODO code related to `sendData` and setting up a server and characteristic. I tried to brush it up but my problem is that even the simple example (without sendData or anything) doesn't work. The advertised service doesn't show up in any scans.

Same here, I pulled up a scanner and never found the advertised service.

juliansteenbakker, 2026-09-03:
It's been a while, but i have released a new version of this package, [v3.0.0](https://github.com/juliansteenbakker/flutter_ble_peripheral/releases/tag/v3.0.0), together with the other side of it, [flutter_ble_central](https://github.com/juliansteenbakker/flutter_ble_central/releases/tag/v1.0.0) v1.0.0. 3.0.0 adds a gatt server with tx and rx characteristics, so a central can connect and read and write over the link. The example app does exactly that, with flutter_ble_central on the other end.

Please check again with these versions, and reopen if it's still there.

closed 2026-09-03 by juliansteenbakker