juliansteenbakker/flutter_ble_peripheral #94: Error when setting value for serviceData

ONLYstcm, 2022-08-28:
When trying to use  serviceData in AdvertiseData, for example:
`serviceData: [7,8,9,10]`

I get the following error at the beginning of a advertisment function:
```
E/MethodChannel#dev.steenbakker.flutter_ble_peripheral/ble_state( 8553): Failed to handle method call
E/MethodChannel#dev.steenbakker.flutter_ble_peripheral/ble_state( 8553): java.lang.ClassCastException: java.util.ArrayList cannot be cast to byte[]
E/MethodChannel#dev.steenbakker.flutter_ble_peripheral/ble_state( 8553): 	at dev.steenbakker.flutter_ble_peripheral.FlutterBlePeripheralPlugin.startPeripheral(FlutterBlePeripheralPlugin.kt:167)
E/MethodChannel#dev.steenbakker.flutter_ble_peripheral/ble_state( 8553): 	at dev.steenbakker.flutter_ble_peripheral.FlutterBlePeripheralPlugin.onMethodCall(FlutterBlePeripheralPlugin.kt:123)
E/MethodChannel#dev.steenbakker.flutter_ble_peripheral/ble_state( 8553): 	at io.flutter.plugin.common.MethodChannel$IncomingMethodCallHandler.onMessage(MethodChannel.java:262)
E/MethodChannel#dev.steenbakker.flutter_ble_peripheral/ble_state( 8553): 	at io.flutter.embedding.engine.dart.DartMessenger.invokeHandler(DartMessenger.java:295)
E/MethodChannel#dev.steenbakker.flutter_ble_peripheral/ble_state( 8553): 	at io.flutter.embedding.engine.dart.DartMessenger.lambda$dispatchMessageToQueue$0$io-flutter-embedding-engine-dart-DartMessenger(DartMessenger.java:319)
E/MethodChannel#dev.steenbakker.flutter_ble_peripheral/ble_state( 8553): 	at io.flutter.embedding.engine.dart.DartMessenger$$ExternalSyntheticLambda0.run(Unknown Source:12)
E/MethodChannel#dev.steenbakker.flutter_ble_peripheral/ble_state( 8553): 	at android.os.Handler.handleCallback(Handler.java:938)
E/MethodChannel#dev.steenbakker.flutter_ble_peripheral/ble_state( 8553): 	at android.os.Handler.dispatchMessage(Handler.java:99)
E/MethodChannel#dev.steenbakker.flutter_ble_peripheral/ble_state( 8553): 	at android.os.Looper.loopOnce(Looper.java:226)
E/MethodChannel#dev.steenbakker.flutter_ble_peripheral/ble_state( 8553): 	at android.os.Looper.loop(Looper.java:313)
E/MethodChannel#dev.steenbakker.flutter_ble_peripheral/ble_state( 8553): 	at android.app.ActivityThread.main(ActivityThread.java:8751)
E/MethodChannel#dev.steenbakker.flutter_ble_peripheral/ble_state( 8553): 	at java.lang.reflect.Method.invoke(Native Method)
E/MethodChannel#dev.steenbakker.flutter_ble_peripheral/ble_state( 8553): 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:571)
E/MethodChannel#dev.steenbakker.flutter_ble_peripheral/ble_state( 8553): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1135)
```

ahmdsdk, 2023-08-21:
Hello @ONLYstcm have you been able to solve this?

khanhnmbk, 2024-09-05:
I got the same error. Anyone resolves it?

darshsingh116, 2025-08-17:
Found a possible workaround. No error came for below code after debugging the package code found out that serviceDataUUID is also necessary.

```
final advertiseData = AdvertiseData(
        serviceUuid: "0000180a-0000-1000-8000-00805f9b34fb",
        serviceDataUuid: "0000180a-0000-1000-8000-00805f9b34fd",
        serviceData: payload,
        includeDeviceName: false,
      );

final advertiseSetParameters = AdvertiseSetParameters(
        connectable: false,
        legacyMode: true,
        scannable: false,
        includeTxPowerLevel: false,
      );

await _peripheral.stop(); // Stop any previous advertising

await _peripheral.start(
  advertiseData: advertiseData,
  advertiseSetParameters: advertiseSetParameters,
);
```

Note that there is still constraint of Advertising size taken into consideration.

juliansteenbakker, 2026-09-03:
It's been a while, but i have released a new version of this package, [v3.0.0](https://github.com/juliansteenbakker/flutter_ble_peripheral/releases/tag/v3.0.0), together with the other side of it, [flutter_ble_central](https://github.com/juliansteenbakker/flutter_ble_central/releases/tag/v1.0.0) v1.0.0. The advertise byte payloads are read without casting to `ByteArray` since, so setting `serviceData` works now.

Please check again with these versions, and reopen if it's still there.

closed 2026-09-03 by juliansteenbakker