juliansteenbakker/flutter_ble_peripheral #108: How to scan AdvertiseData ble.

bondPiyapan, 2022-11-03:
i'm use flutter_blue scan ble not found AdvertiseData.

Example : `final AdvertiseData advertiseData = AdvertiseData(
      serviceUuid: 'bf27730d-860a-4e09-889c-2d8b6a9e0fe7',
      manufacturerId: 1234,
      manufacturerData: Uint8List.fromList([1, 2, 3, 4, 5, 6]),
      localName: 'TESTBLE',
      serviceDataUuid: 'TESTTEST')`

Thank you.

MrCsabaToth, 2022-12-29:
Are you using flutter_blue_peripherial and flutte_blue in the same project? Your description is very terse, could you share some more code or give more details? Flutter blue doesn't seem to be maintained BTW, have you tried flutter_blue_plus by boskosg?

juliansteenbakker, 2026-09-03:
It's been a while, but i have released a new version of this package, [v3.0.0](https://github.com/juliansteenbakker/flutter_ble_peripheral/releases/tag/v3.0.0), together with the other side of it, [flutter_ble_central](https://github.com/juliansteenbakker/flutter_ble_central/releases/tag/v1.0.0) v1.0.0. 3.0.0 splits the advertise data per platform, and flutter_ble_central v1.0.0 is the scanning side if you want both ends on the same api.

Please check again with these versions, and reopen if it's still there.

closed 2026-09-03 by juliansteenbakker