juliansteenbakker/flutter_ble_peripheral #113: The advertised service doesn't show up in any scans

MrCsabaToth, 2022-12-31:
I experimented with the example app. Unfortunately I couldn't get it to work: the advertised service doesn't show up in any scans. I'm using a OnePlus Nord Android phone with Android 12 which supposedly supports peripheral mechanisms. The example code doesn't give me any error messages, Bluetooth is enabled and the app has permissions. I tried both the `Advertise` method and the `AdvertiseSet`. I'm trying to scan the BLE peripheral with other Android phones using nRF Connect, an iPad with nRF Connect and my laptop (Linux with BlueZ).

```
  final AdvertiseData advertiseData = AdvertiseData(
    // FTMS
    serviceUuid: '00001826-0000-1000-8000-00805f9b34fb',
    // Indoor Bike
    serviceDataUuid: '00002ad2-0000-1000-8000-00805f9b34fb',
    // according to zwifit/src/ble/enable.js buildAdvertisingData
    serviceData: Uint8List.fromList([2, 0x01, 0x05, 3, 0x03, 0x18, 0x26]),
    // Precor
    manufacturerId: 1819,
    // By the FTMS specification chapter 3
    manufacturerData: Uint8List.fromList([0, 0x18, 0x26, 1, 32, 0]),
  );

  final AdvertiseSettings advertiseSettings = AdvertiseSettings(
    connectable: true,
    advertiseMode: AdvertiseMode.advertiseModeBalanced,
    txPowerLevel: AdvertiseTxPower.advertiseTxPowerMedium,
    timeout: 3600,
  );

  final AdvertiseSetParameters advertiseSetParameters = AdvertiseSetParameters(
    // connectable: true,
    scannable: true,
    txPowerLevel: txPowerMedium,
    duration: 3600,
  );
```

It's another issue that if I'd see that service I don't seem to find a way to actually communicate data, which kinda makes the whole experiment mute: it has no value if I cannot communicate (#85). But even the mute service does not seem to show.

microgeniusz80, 2023-09-03:
I/BluetoothLeAdvertiser(15672): startAdvertisingSet, calling package name = dev.steenbakker.flutter_ble_peripheral_example
I/BluetoothAdapter(15672): isLeEnabled(): ON
I/FlutterBlePeripheral(15672): onStartSuccess() mode: 2, txPOWER 1
I/BLE Peripheral state (15672): advertising
I/BluetoothLeAdvertiser(15672): stopAdvertising is called
I/BluetoothLeAdvertiser(15672): stopAdvertisingSet, calling package name = dev.steenbakker.flutter_ble_peripheral_example

it seems that as soon as I/BLE Peripheral state (15672): advertising happened, within a second: I/BluetoothLeAdvertiser(15672): stopAdvertising is called

causing, it not to show any advertisement

juliansteenbakker, 2026-09-03:
It's been a while, but i have released a new version of this package, [v3.0.0](https://github.com/juliansteenbakker/flutter_ble_peripheral/releases/tag/v3.0.0), together with the other side of it, [flutter_ble_central](https://github.com/juliansteenbakker/flutter_ble_central/releases/tag/v1.0.0) v1.0.0. Android service uuids not broadcasting and the advertisement payload were both fixed since, so this should show up in a scan now.

Please check again with these versions, and reopen if it's still there.

closed 2026-09-03 by juliansteenbakker