juliansteenbakker/flutter_ble_peripheral #118: Establish the connection

oshrimaayan, 2023-01-29:
After advertising, the other app is trying to connect but doesn't get a response.
Can you please document this step?

How do I establish the connection and start to move data between the apps?

zhuwei, 2023-11-02:
I have the same question, doesn't this library have this feature? What's the use of that?

ekuleshov, 2023-12-17:
To allow incoming connection it has to provide a GATT server. E.g. see https://github.com/juliansteenbakker/flutter_ble_peripheral/pull/155

juliansteenbakker, 2026-09-03:
It's been a while, but i have released a new version of this package, [v3.0.0](https://github.com/juliansteenbakker/flutter_ble_peripheral/releases/tag/v3.0.0), together with the other side of it, [flutter_ble_central](https://github.com/juliansteenbakker/flutter_ble_central/releases/tag/v1.0.0) v1.0.0. 3.0.0 adds a gatt server with tx and rx characteristics, so a central can connect and read and write over the link. The example app does exactly that, with flutter_ble_central on the other end.

Please check again with these versions, and reopen if it's still there.

closed 2026-09-03 by juliansteenbakker