juliansteenbakker/flutter_ble_peripheral #144: Use as HID device

taavirubenhagen, 2023-08-07:
I can't find any documentation about the actual data exchange between the devices, only about the advertising. Is the actual functionality that would be needed to mimic an extremely minimalist, yet complete bluetooth device on Android available yet, and if yes, how can I read and write data?
Ideally, the app that I'm building should be able to turn my phone into a PC remote control device, but really any kind of data exchange could help for a workaround.

rohitsangwan01, 2023-08-12:
Hey @taavirubenhagen have you find any solution, i was also looking to do something similar for some hid mouse related task, so created a sdk, its not polished yet, but you can give [this](https://github.com/rohitsangwan01/ble_peripheral) a try,

taavirubenhagen, 2023-08-15:
I decided to switch to WiFi communication instead since it is easier in general, but this looks exactly like what I had been searching for, thank you

closed 2023-08-15 by taavirubenhagen