juliansteenbakker/flutter_ble_peripheral #150: stopAdvertising is automatically called, within 0.5second of advertisement started

microgeniusz80, 2023-09-03:
latest library update causes advertisement not showing anymore.

I/BluetoothLeAdvertiser(15672): startAdvertising is called
I/BluetoothAdapter(15672): isLeEnabled(): ON
I/BluetoothLeAdvertiser(15672): startAdvertisingSet, calling package name = dev.steenbakker.flutter_ble_peripheral_example
I/BluetoothAdapter(15672): isLeEnabled(): ON
I/FlutterBlePeripheral(15672): onStartSuccess() mode: 2, txPOWER 1
I/BLE Peripheral state (15672): advertising
I/BluetoothLeAdvertiser(15672): stopAdvertising is called
I/BluetoothLeAdvertiser(15672): stopAdvertisingSet, calling package name = dev.steenbakker.flutter_ble_peripheral_example
V/AudioManager(15672): querySoundEffectsEnabled...
I/BluetoothLeAdvertiser(15672): stopAdvertising is called

it seems that as soon as I/BLE Peripheral state (15672): advertising happened, within a second: I/BluetoothLeAdvertiser(15672): stopAdvertising is called.

I/BLE Peripheral state (15672): advertising
I/BluetoothLeAdvertiser(15672): stopAdvertising is called
These two lines were generated together when the start advertising button was clicked.

causing, it not to show any advertisement

previous version was fine

aleadag, 2023-12-05:
same issue here, any suggestion?

karthicbz, 2023-12-19:
set duration in AdvertiseSettings
final AdvertiseSettings advertiseSettings = AdvertiseSettings(timeout: 50000);

karthicbz, 2023-12-19:
> same issue here, any suggestion?

set duration in AdvertiseSettings
final AdvertiseSettings advertiseSettings = AdvertiseSettings(timeout: 50000);

stam0912, 2023-12-21:
Are you using android? The timeout is 400ms by default. You need to set the timeout value if you want it advertise longer.

"Limit advertising to a given amount of time. May not exceed 180000 milliseconds. Default: 400 milliseconds"

juliansteenbakker, 2026-09-03:
It's been a while, but i have released a new version of this package, [v3.0.0](https://github.com/juliansteenbakker/flutter_ble_peripheral/releases/tag/v3.0.0), together with the other side of it, [flutter_ble_central](https://github.com/juliansteenbakker/flutter_ble_central/releases/tag/v1.0.0) v1.0.0. The advertising state handling on android was reworked since, including sending the results from the main thread.

Please check again with these versions, and reopen if it's still there.

closed 2026-09-03 by juliansteenbakker