juliansteenbakker/flutter_ble_peripheral #151: Unhandled Exception: MissingPluginException(requestPermissions)

RubenGarcia, 2023-09-04:
BluetoothPeripheralState peripheralPermission = await flutterBlePeripheral.requestPermission();

throws this, although the permission is granted. Please investigate.
```
E/flutter ( 2983): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: MissingPluginException(No implementation found for method requestPermissions on channel dev.steenbakker.flutter_ble_peripheral/ble_state)
E/flutter ( 2983): #0      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:308:7)
E/flutter ( 2983): <asynchronous suspension>
E/flutter ( 2983): #1      FlutterBlePeripheral.requestPermission (package:flutter_ble_peripheral/src/flutter_ble_peripheral.dart:135:9)
E/flutter ( 2983): <asynchronous suspension>
```

RubenGarcia, 2023-09-06:
Another comment.
https://pub.dev/packages/flutter_ble_peripheral/example
uses always
await FlutterBlePeripheral().XXX
and does not save the result of  FlutterBlePeripheral() into a variable for later use.
Is saving it into a variable forbidden?

RubenGarcia, 2023-09-06:
Changing the code to use FlutterBlePeripheral() everywhere did not change behaviour.

RubenGarcia, 2023-09-06:
Workaround: use
import 'package:permission_handler/permission_handler.dart';
to request the permissions explicitly.

juliansteenbakker, 2026-09-03:
It's been a while, but i have released a new version of this package, [v3.0.0](https://github.com/juliansteenbakker/flutter_ble_peripheral/releases/tag/v3.0.0), together with the other side of it, [flutter_ble_central](https://github.com/juliansteenbakker/flutter_ble_central/releases/tag/v1.0.0) v1.0.0. The permission handling was rewritten since 1.x.

Please check again with these versions, and reopen if it's still there.

closed 2026-09-03 by juliansteenbakker