juliansteenbakker/flutter_ble_peripheral #158: cannot receive state idle callback

Shik1266, 2023-10-12:
@juliansteenbakker 
i cannot receive state idle callback when i stop advertising with onPeripheralStateChanged

stam0912, 2023-12-21:
Do you have a solution yet? Is it a bug on android only? I don't receive the statechanged callback after timeout/stop advertising, but ios works.

juliansteenbakker, 2026-09-03:
It's been a while, but i have released a new version of this package, [v3.0.0](https://github.com/juliansteenbakker/flutter_ble_peripheral/releases/tag/v3.0.0), together with the other side of it, [flutter_ble_central](https://github.com/juliansteenbakker/flutter_ble_central/releases/tag/v1.0.0) v1.0.0. The peripheral state reporting was reworked since, so idle should come through when you stop.

Please check again with these versions, and reopen if it's still there.

closed 2026-09-03 by juliansteenbakker