juliansteenbakker/flutter_ble_peripheral #160: PlatformException (PlatformException(18, UNDOCUMENTED, startAdvertising, null))

zhuwei, 2023-10-31:
Setting `includeDeviceName: true` on Android will result in an exception

SyedKaz00, 2023-11-03:
Where is this setting located, also is there a fix?

CellCS, 2024-03-27:
If add "includeDeviceName: true", here is using Android 12.

```
  final AdvertiseData advertisePeriodicData = AdvertiseData(
      localName: "bletester",
      includeDeviceName: true);

```
This exception log:

```
I/BluetoothLeAdvertiser(22345): startAdvertising is called
I/BLE Peripheral state (22345): idle
E/flutter (22345): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(2, ADVERTISE_FAILED_TOO_MANY_ADVERTISERS, startAdvertising, null)
E/flutter (22345): #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:651:7)
E/flutter (22345): #1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:334:18)
E/flutter (22345): <asynchronous suspension>
E/flutter (22345): #2      FlutterBlePeripheral.start (package:flutter_ble_peripheral/src/flutter_ble_peripheral.dart:89:9)
E/flutter (22345): <asynchronous suspension>
E/flutter (22345): #3      FlutterBlePeripheralExampleState.build.<anonymous closure> (package:mybleperitest/main.dart:172:19)
E/flutter (22345): <asynchronous suspension>
E/flutter (22345): 
```

Ayesha-Sad-Hours, 2025-01-28:
im having the same issue. did u find a fix?

amitasdf, 2025-02-11:
im having the same issue. did u find a fix?

zhuwei, 2025-02-11:
im having the same issue. did u find a fix?

CellCS, 2025-02-11:
Sorry,forget whether this issue is fixed or not,   i turned to use ''flutter_blue_plus'' package,