juliansteenbakker/flutter_ble_peripheral #205: [flutter_ble_peripheral] didStartAdvertising: Error "Advertising has already started" on iOS

Kat-Whitehouse, 2024-10-02:
I have picked up the development on an app which uses a phone to send BLE data to an IOT device.  The app works as it should on Android, but as we all know, it works completely differently on iOS.

I can only seem to send one advertiseData packet and then the iPhone just continues to send it.  I have a sniffer (using Tera Term) and so I can see what is being sent.  If I then send another advertiseData packet, it is just ignored by the IOT device.  I cannot find a way of stopping the iPhone from advertising so that I can send the new packet to the device.  The .stop() doesn't seem to do anything and the iPhone just continuously advertises the original advertiseData packet.

Does anyone have any ideas?  How do I use .stop() please?  Incidentally, the Android code does not need to use .stop().  When I send new advertiseData, it just does it and the device responds as it should.

huzaifa846, 2024-10-02:
facing same issue

juliansteenbakker, 2026-09-03:
It's been a while, but i have released a new version of this package, [v3.0.0](https://github.com/juliansteenbakker/flutter_ble_peripheral/releases/tag/v3.0.0), together with the other side of it, [flutter_ble_central](https://github.com/juliansteenbakker/flutter_ble_central/releases/tag/v1.0.0) v1.0.0. As you found, `stop()` before `start()`. 3.0.0 also returns the advertising state from `start` on every platform, and keeps `isAdvertising` true while a central is connected.

Please check again with these versions, and reopen if it's still there.

closed 2026-09-03 by juliansteenbakker